Seeed-Studio/Seeed_Arduino_SSCMA #5: Grove GPS Air530 with XIAO-ESP32S3 Bug?

hpssjellis, 2024-04-11:
Hi @LynnL4 I am trying to get the Grove GPS Air530 to work with the XIAO-ESP32S3 but the use of softserial is causing me some issues.

Please move this to the correct repo if there is a better location.

The ordinal Grove GPS code is and works fine on the XIAO-SAMD

```
#include <SoftwareSerial.h>
SoftwareSerial SoftSerial(2, 3);
unsigned char buffer[64];                   // buffer array for data receive over serial port
int count=0;                                // counter for buffer array
void setup()
{
    SoftSerial.begin(9600);                 // the SoftSerial baud rate
    Serial.begin(9600);                     // the Serial port of Arduino baud rate.
}

void loop()
{
    if (SoftSerial.available())                     // if date is coming from software serial port ==> data is coming from SoftSerial shield
    {
        while(SoftSerial.available())               // reading data into char array
        {
            buffer[count++]=SoftSerial.read();      // writing data into array
            if(count == 64)break;
        }
        Serial.write(buffer,count);                 // if no data transmission ends, write buffer to hardware serial port
        clearBufferArray();                         // call clearBufferArray function to clear the stored data from the array
        count = 0;                                  // set counter of while loop to zero 
    }
    if (Serial.available())                 // if data is available on hardware serial port ==> data is coming from PC or notebook
    SoftSerial.write(Serial.read());        // write it to the SoftSerial shield
}

void clearBufferArray()                     // function to clear buffer array
{
    for (int i=0; i<count;i++)
    {
        buffer[i]=NULL;
    }                      // clear all index of array with command NULL
}
```

.

.

 Without softserial but just using UART this is my attempt:

```
//#include <SoftwareSerial.h>
//SoftwareSerial SoftSerial(2, 3);
unsigned char buffer[64];                   // buffer array for data receive over serial port
int count=0;                                // counter for buffer array
void setup()
{
    Serial1.begin(9600);                 // the SoftSerial baud rate
    Serial.begin(115200);                     // the Serial port of Arduino baud rate.
}

void loop()
{
    if (Serial1.available())                     // if date is coming from software serial port ==> data is coming from SoftSerial shield
    {
        while(Serial1.available())               // reading data into char array
        {
            buffer[count++]=Serial1.read();      // writing data into array
            if(count == 64)break;
        }
        Serial.write(buffer,count);                 // if no data transmission ends, write buffer to hardware serial port
        clearBufferArray();                         // call clearBufferArray function to clear the stored data from the array
        count = 0;                                  // set counter of while loop to zero 
    }
    if (Serial.available())                 // if data is available on hardware serial port ==> data is coming from PC or notebook
    Serial1.write(Serial.read());        // write it to the SoftSerial shield
}

void clearBufferArray()                     // function to clear buffer array
{
    for (int i=0; i<count;i++)
    {
        buffer[i]=NULL;
    }                      // clear all index of array with command NULL
}

```

but not getting anything useful, can you make a suggestion for a better library that would work with the ESP32 or give me some suggestions for how to get this GPS working with the XIAO-ESP32S3?

Lesords, 2024-10-18:
Hello,

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

hpssjellis, 2024-10-19:
> Hello,
> 
> I'm very sorry to have kept you waiting so long.
> 
> Do you still have this problem now?

Hi @Lesords  no worries. I will be working on the curriculum again this winter as I prepare for using the meshtastic LoRa XIAO esp32s3. So anytime you have any suggestions. I still need to connect to the GPS, but I also need to get useful data from the GPS. Normally GPS modules just give the raw data. I am hoping we can find a way to parse the data to just give the longnitude and latitude ( and height above sea level  if possible)

Lesords, 2024-11-08:
I don't have this hardware device at the moment, I will verify it once I have it and let you know the result

closed 2024-12-24 by LynnL4