Seeed-Studio/Seeed_Arduino_SSCMA #18: XIAO ESP32C6 -  Processor or architecture not support!

just4give, 2024-05-05:
I am trying to compile camera_web_server example for XIAO ESP32C6 but getting below error. Looks like `ARDUINO_ARCH_ESP32` is not yet coded in Seeed_Arduino_FreeRTOS library? Can you please guide? 

```
In file included from /Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:65,
                 from /Users/mithundas/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-rc1/cores/esp32/Arduino.h:33,
                 from /private/var/folders/fm/x0fh_qqj3954m97bjgctrlwm0000gn/T/arduino/sketches/C444280EC8F7A4E9A576314394E7FF84/sketch/grove_vision_ai_camera_web_server.ino.cpp:1:
/Users/mithundas/Documents/Arduino/libraries/Seeed_Arduino_FreeRTOS/src/FreeRTOSConfig.h:32:2: error: #error "Processor or architecture not support!"
   32 | #error "Processor or architecture not support!"
      |  ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:90:6: error: #error Missing definition: configMINIMAL_STACK_SIZE must be defined in FreeRTOSConfig.h. configMINIMAL_STACK_SIZE defines the size (in words) of the stack allocated to the idle task. Refer to the demo project provided for your port for a suitable value.
   90 |     #error Missing definition:  configMINIMAL_STACK_SIZE must be defined in FreeRTOSConfig.h.  configMINIMAL_STACK_SIZE defines the size (in words) of the stack allocated to the idle task.  Refer to the demo project provided for your port for a suitable value.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:94:6: error: #error Missing definition: configMAX_PRIORITIES must be defined in FreeRTOSConfig.h. See the Configuration section of the FreeRTOS API documentation for details.
   94 |     #error Missing definition:  configMAX_PRIORITIES must be defined in FreeRTOSConfig.h.  See the Configuration section of the FreeRTOS API documentation for details.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:98:6: error: #error configMAX_PRIORITIES must be defined to be greater than or equal to 1.
   98 |     #error configMAX_PRIORITIES must be defined to be greater than or equal to 1.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:102:6: error: #error Missing definition: configNUM_CORES must be defined in FreeRTOSConfig.h
  102 |     #error Missing definition:  configNUM_CORES must be defined in FreeRTOSConfig.h
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:106:6: error: #error configNUM_CORES must be defined to either 1 or 2.
  106 |     #error configNUM_CORES must be defined to either 1 or 2.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:110:6: error: #error Missing definition: configUSE_PREEMPTION must be defined in FreeRTOSConfig.h as either 1 or 0. See the Configuration section of the FreeRTOS API documentation for details.
  110 |     #error Missing definition:  configUSE_PREEMPTION must be defined in FreeRTOSConfig.h as either 1 or 0.  See the Configuration section of the FreeRTOS API documentation for details.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:114:6: error: #error Missing definition: configUSE_IDLE_HOOK must be defined in FreeRTOSConfig.h as either 1 or 0. See the Configuration section of the FreeRTOS API documentation for details.
  114 |     #error Missing definition:  configUSE_IDLE_HOOK must be defined in FreeRTOSConfig.h as either 1 or 0.  See the Configuration section of the FreeRTOS API documentation for details.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:118:6: error: #error Missing definition: configUSE_TICK_HOOK must be defined in FreeRTOSConfig.h as either 1 or 0. See the Configuration section of the FreeRTOS API documentation for details.
  118 |     #error Missing definition:  configUSE_TICK_HOOK must be defined in FreeRTOSConfig.h as either 1 or 0.  See the Configuration section of the FreeRTOS API documentation for details.
      |      ^~~~~
/Users/mithundas/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:122:6: error: #error Missing definition: configUSE_16_BIT_TICKS must be defined in FreeRTOSConfig.h as either 1 or 0. See the Configuration section of the FreeRTOS API documentation for details.
  122 |     #error Missing definition:  configUSE_16_BIT_TICKS must be defined in FreeRTOSConfig.h as either 1 or 0.  See the Configuration section of the FreeRTOS API documentation for details.
      |      ^~~~~
exit status 1

Compilation error: exit status 1
```

Pillar1989, 2024-05-07:
@just4give  As you can see, Seeed_Arduino_FreeRTOS does not support Seeed_Arduino_FreeRTOS. But esp Arduino already includes freertos, so you can still use the functionality of the rtos.

closed 2024-05-07 by Pillar1989

just4give, 2024-05-08:
@Pillar1989 I am sorry but it's not clear to me. Are you saying I should remove the `Seeed_Arduino_FreeRTOS` library ?

just4give, 2024-05-14:
For something visiting this page in future - at the moment this library supports XIAO ESP32 S3. If you got C3 or C6, probably they won't work.