Seeed-Studio/Seeed_Arduino_SSCMA #24: [camera web server example] FreeRTOS.h build error w/ Xiao ESP32C3

mbz4, 2024-06-02:
**Describe the bug**
Trouble building camera web server example - missing library reference: FreeRTOS.h 
Not sure which FreeRTOS.h to use, there's so many and its very configurable.

*Observations:*
- Suspect the C3 might be missing a (Seeed) FreeRTOS.h support package.
- The C3 may not have sufficient RAM for this example.
  - Would the Xiao ESP32-S3 Sense work better in its place (w/ Grove Vision Module v2)?

**To Reproduce**
Hardware: Xiao ESP32C3 + Grove Vision Module v2

Steps to reproduce the behavior:
1. Configure project w/ PlatformIO or ArduinoIDEv2
2. Run Build / Compile Web Camera Server example project 

**Expected behavior**
Expected successful build, build errors instead.

**Platformio.ini file:** 
[env:seeed_xiao_esp32c3]
platform = espressif32
board = seeed_xiao_esp32c3
board_build.mcu = esp32c3
board_build.f_cpu = 160000000L
monitor_speed = 115200
upload_protocol = esptool
framework = arduino
lib_deps = 
    seeed-studio/Seeed_Arduino_SSCMA@^1.0.0
    bblanchon/ArduinoJson@^7.0.4
    ; linlin-study/FreeRTOS-Kernel@^10.4.4-1 (tried several, not sure which to use here)
    hideakitai/ArduinoEigen@^0.3.2

**Build Output**

Processing seeed_xiao_esp32c3 (platform: espressif32; board: seeed_xiao_esp32c3; framework: arduino)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/seeed_xiao_esp32c3.html
PLATFORM: Espressif 32 (6.4.0) > Seeed Studio XIAO ESP32C3
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Seeed_Arduino_SSCMA @ 1.0.0
|-- ArduinoJson @ 7.0.4
|-- ArduinoEigen @ 0.3.2
|-- Wire @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/seeed_xiao_esp32c3/src/BYTETracker.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/src/STrack.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/src/app_httpd.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/src/kalmanFilter.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/src/lapjv.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/src/main.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/src/utils.cpp.o
Building .pio/build/seeed_xiao_esp32c3/bootloader.bin
Generating partitions .pio/build/seeed_xiao_esp32c3/partitions.bin
esptool.py v4.5.1
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
src/app_httpd.cpp:21:10: fatal error: FreeRTOS.h: No such file or directory

******************************************************************
* Looking for FreeRTOS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FreeRTOS.h"
* Web  > https://registry.platformio.org/search?q=header:FreeRTOS.h
*
******************************************************************

 #include <FreeRTOS.h>
          ^~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/seeed_xiao_esp32c3/lib80c/SPI/SPI.cpp.o
*** [.pio/build/seeed_xiao_esp32c3/src/app_httpd.cpp.o] Error 1
=========================================== [FAILED] Took 14.22 seconds ===========================================

mbz4, 2024-06-03:
Updates:
- Concluded C3 not supported
  - Need S3 instead
- assume the camera web server example deploys to the xiao not grove
- not sure what to deploy to the vision module 
- vision module v2 wiki shows WIP on image collection script:
  - https://wiki.seeedstudio.com/grove_vision_ai_v2_sscma/ ...

> We highly recommend that you use Grove Vision AI V2 to take pictures of your dataset, which is best for Grove Vision AI V2. Grove Vision AI V2 is currently developing the functionality part of taking pictures, and once that is done, you can use Grove Vision AI V2 to capture pictures for your dataset. Until then, you can use photos taken by other devices as datasets.

Figured to try collect the images using the xiao x vision module combo with this camera server example.
Will try deploying and get this example running again.
Ultimately just need the image dataset...

mbz4, 2024-06-04:
updates:
- build error does not repeat with xiao esp32s3 
  - uploads successfully but stuck in reboot loop if having serial console open 
  - browser page not loading

closed 2024-06-20 by mbz4

Muhammad-Yunus, 2024-08-31:
Hi guys,

I'm able to build camera webserver example for xiao esp32 c3 after commenting #include <FreeRTOS.h> in app_httpd.cpp. 
[image]