Seeed-Studio/Seeed_Arduino_SSCMA #25: [Improve request] Change AI.boxes[n].target value type. From uint_8 to char array.

djairjr, 2024-06-12:
It would be interesting if the values ​​of AI.boxes.target resulted in a char array, instead of uint_8. For object classification, this functionality would make things much easier. I believe it should reflect a change in the Groove Ai firmware that would also send data in this format (char array) instead of uint_8.

On Seeed_Arduino_SSCMA.h, line 149:
typedef struct
{
    uint16_t x;
    uint16_t y;
    uint16_t w;
    uint16_t h;
    uint8_t score;
    uint8_t target; // here char *target;
} boxes_t;

I believe this implementation would also be interesting for other structures.

LynnL4, 2024-11-04:
Hi, I'm planning to add support for the global.gc option. Thank you!

LynnL4, 2024-12-26:
FYI https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/tree/main/examples/inference_class

closed 2024-12-26 by LynnL4