Seeed-Studio/Seeed_Arduino_SSCMA #28: Camera Web Server Issues XIAO ESP32S3 Sense + Grove Vision AI Module with OV5647-160

rashawnoverton, 2024-07-30:
I tried to use the camera web server example and ran into a few errors.

1838032971, 2024-08-17:
Hi, I have met the same question. No errors for me but I just can't start the stream. It just show me the two warning.
[image]
[image]

Lesords, 2024-11-04:
Hello,

I\'m going to close this issue, feel free to re-open it if you have any other questions.

[Autocommentary]

idogolds, 2024-11-05:
Hey,

I’d like to reopen this issue as I’m still experiencing the same problem and have verified each of the points mentioned previously.

PSRAM Status: I have confirmed that PSRAM is enabled on my ESP32.
Dependency Versions: All dependent packages, including the Arduino ESP32 package, are fully up to date.
Connection Details: The ESP32S3 is physically connected to the WE2 module, and I’ve verified that the OV5647-160 camera is connected properly to the Grove Vision 2 as well. However, I'm not entirely certain about the software linkage.
Regarding Logs: I'm currently unsure about how to retrieve the software logs for my ESP32S3 setup with the Grove Vision 2.

Lesords, 2024-11-06:
> Hey,
>
> I’d like to reopen this issue as I’m still experiencing the same problem and have verified each of the points mentioned previously.
>
> PSRAM Status: I have confirmed that PSRAM is enabled on my [...]

@iChizer0 Help with this one

idogolds, 2024-11-12:
Hey @iChizer0   , thanks for the help! Out of all the supported libraries, it turned out that I only had the ESP32 library version wrong. I was using 3.0.7 instead of 2.0.14, so I’ve now updated it as recommended. However, I’m now running into a new issue where the web server doesn’t load at all. Here’s the output log I’m seeing:

Rebooting...
ESP-ROM:esp32s3-20210327
Build: Mar 27 2021
rst:0xc (RTC_SW_CPU_RST), boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4208c0fe
SPIWP:0xee
mode: DIO, clock div:1
load: 0x3fce3808, len: 0x44c
load: 0x403c9700, len: 0xbd8
load: 0x403cc700, len: 0x2a80
entry 0x403c98d0
E (176) esp_core_dump_flash: No core dump partition found!
E (176) esp_core_dump_flash: No core dump partition found!
file: "/home/ido/.arduino15/packages/esp32/hardware/esp32/2.0.14/cores/esp32/esp32-hal-uart.c" line 212
func: uartBegin
expression: uart_driver_install(uart_nr, rx_buffer_size, tx_buffer_size, 20, &(uart->uart_event_queue), 0
abort() was called at PC 0x4037d6f3 on core 1

Backtrace: 0x403777ae:0x3fcebde0 0x4037d6fd:0x3fcebe00 0x4038361d:0x3fcebe20 0x4037d6f3:0x3fcebea0 0x42015bf3:0x3fcebec0 0x42013aaf:0x3fcebf20 0x420054ea:0x3fcebf70 0x42008d0b:0x3fcebfa0 0x42015cde:0x3fcebfd0

ELF file SHA256: edffd004e9bfb342

idogolds, 2024-12-05:
It works on I2C ,It successfully starts, but after displaying a few frames with a low frame rate, it pauses. here's the log:
11:27:25.597 -> [ 43031][I][app_httpd.cpp:257] proxyCallback(): Received 4659 bytes...
11:27:25.597 -> [ 43033][I][app_httpd.cpp:254] proxyCallback(): Discarded 1 old responses...
11:27:25.597 -> [ 43039][I][a[p p4_3h0t4t0p]d[.E]c[papp:2p5_h7t]t pprd.ocxypCpa:l7l41ba] cskt(r):e aRme_creeisvueldt_ h4a4n6d9 lbeyrt()es: .S.e.n
11:27:25.597 -> d results failed...
11:27:26.145 -> [ 43612][I][app_httpd.cpp:254] proxyCallback(): Discarded 1 old responses...
11:27:26.178 -> [ 43613][I][app_httpd.cpp:257] proxyCallback(): Received 4469 bytes...
11:27:26.178 -> [ 43616][I][app_httpd.cpp:254] proxyCallback(): Discarded 1 old responses...
11:27:26.178 -> [ 43621][I][app_httpd.cpp:257] proxyCallback(): Received 4671 bytes...
11:27:26.178 -> [ 43629][I][app_httpd.cpp:254] proxyCallback(): Discarded 1 old responses...
v11:27:27.184 -> [ 44620][I][app_httpd.cpp:257] proxyCallback(): Received 4835 bytes...
11:27:27.184 -> [ 44628][I][app_httpd.cpp:254] proxyCallback(): Discarded 1 old responses...
11:27:27.184 -> [ 44633][I][app_httpd.cpp:257] proxyCallback(): Received 4819 bytes...
11:28:08.711 -> [ 86154][I][app_httpd.cpp:254] proxyCallback(): Discarded 1 old responses...
11:28:08.711 -> [ 86155][I][app_httpd.cpp:257] proxyCallback(): Received 24939 bytes...
11:28:08.711 -> [ 86160][E][app_httpd.cpp:625] stream_result_handler(): Failed to parse json...
11:28:08.711 -> 
{"type": 1, "name": "HTTPD0000A421@INVOKE", "code": 0, "data": {"count": 398, "perf": [2, 33, 0], "boxes": [], "resolution": [240, 240], "image":...

LynnL4, 2024-12-24:
@idogolds 
Which board are you using, and how long do you expect it to run before hang out?

LynnL4, 2024-12-27:
@idogolds 
Thank you for your feedback. I have successfully reproduced the issue you mentioned and have fixed the related problem. It is currently undergoing internal testing, and once the tests are completed successfully, I will upload it to the SenseCraft AI platform.
https://github.com/Seeed-Studio/sscma-example-we2/releases/tag/20241226

closed 2025-01-17 by github-actions[bot]