Seeed-Studio/Seeed_Arduino_SSCMA #34: Xiao ESP32 C3 on Grove Vision AI V2 (Arduino IDE issue)

s2244521, 2024-12-20:
Hi all,

I use Xiao ESP32 C3 on Grove Vision AI V2 to inference and get inference results through Arduino IDE but after compiling, I think the code is written to  the flash successfully but the process stops and the following fig shows the error message. How to solve this issue and what is the issue, the Timer or WatchDog or .....?
My code is example of Seeed Arduino SSCMA infernce. 

[image]

LynnL4, 2024-12-20:
Il tuo grove visision ai v2 funziona correttamente e qual è il modo in cui hai collegato lo xiao C3 al grove vision ai v2 sul sito https://sensecraft.seeed.cc/ai?

LynnL4, 2024-12-20:
Hi,
Thank you for reaching out. Regarding your question about the Grove Vision AI V2, could you please confirm if it is working properly for you? Additionally, on the [SenseCraft website](https://sensecraft.seeed.cc/ai), how did you connect the Xiao C3 to the Grove Vision AI V2?

closed 2025-01-06 by LynnL4