Seeed-Studio/Seeed_Arduino_SSCMA #36: Differ by Result button is not logically

imnotwannafire, 2025-05-01:
**Describe the bug**
In Camera web server example, when "Differ by Result " is checked. The working mode is SAMPLE. Which mean, AI module will not detect object and not send inference result anymore? So not Differ by result mode will not take any effect.

LynnL4, 2025-05-01:
The original purpose of "differ" is to reduce bandwidth by only outputting results when the detection differs from the previous one. What hardware are you using? I can try to reproduce your situation and see if there's an issue.

imnotwannafire, 2025-05-05:
> The original purpose of "differ" is to reduce bandwidth by only outputting results when the detection differs from the previous one. What hardware are you using? I can try to reproduce your situation and see if there's an issue.

I'm using Grove AI vision v2. 
You can check in source code of index.html. No need to have a real module

LynnL4, 2025-05-06:
Sorry, I didn’t quite understand your point—could you please help clarify what the issue is?

I see the `switch` is defined here:
[https://github.com/Seeed-Studio/Seeed\_Arduino\_SSCMA/blob/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69/examples/camera\_web\_server/web/index.html#L590](https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/blob/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69/examples/camera_web_server/web/index.html#L590)

It eventually affects this part of the code:
[https://github.com/Seeed-Studio/Seeed\_Arduino\_SSCMA/blob/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69/examples/camera\_web\_server/web/index.html#L1747](https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/blob/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69/examples/camera_web_server/web/index.html#L1747)

From what I can tell, it seems to be working as expected.
Let me know if I missed something—I'm happy to take another look.

imnotwannafire, 2025-05-07:
> Sorry, I didn’t quite understand your point—could you please help clarify what the issue is?
> 
> I see the `switch` is defined here:
> 
> [Seeed_Arduino_SSCMA/examples/camera_web_server/web/index.html](https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/blob/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69/examples/camera_web_server/web/index.html#L590)
> 
> Line 590 in [4fe22f2](/Seeed-Studio/Seeed_Arduino_SSCMA/commit/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69)
> 
>  <label for="differed-by-results">Differed by Results</label> 
> It eventually affects this part of the code:
> 
> [Seeed_Arduino_SSCMA/examples/camera_web_server/web/index.html](https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/blob/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69/examples/camera_web_server/web/index.html#L1747)
> 
> Line 1747 in [4fe22f2](/Seeed-Studio/Seeed_Arduino_SSCMA/commit/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69)
> 
>  const cmd = `INVOKE=-1,${isDiffered},0`; 
> From what I can tell, it seems to be working as expected. Let me know if I missed something—I'm happy to take another look.

Yes, could you check this snip code
differedByResults.onchange = () => {
                if (differedByResults.checked) {
                    workingMode = Mode.SAMPLE;
                    sampleOnly.checked = false;
                    disable(sampleSwitch);
                } else {
                    workingMode = Mode.INVOKE;
                    enable(sampleSwitch);
                }
            };
when differeByResults switch is checked, it change workingMode to Sample

LynnL4, 2025-05-07:
Thank you very much for pointing out the issue. I have already fixed it.
https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/commit/b17e2dc5e117e4136bfab772a75a09f1268efb35

closed 2025-05-29 by github-actions[bot]