Seeed-Studio/Seeed_Arduino_SSCMA #37: The inference programme runs incorrectly on Wio Terminal.

limengdu, 2025-05-06:
**Describe the bug**

The sample inference_uart program cannot be successfully compiled on the Wio Terminal.

**To Reproduce**
Steps to reproduce the behavior:

When opening the inference_uart code in Arduino IDE, selecting Wio Terminal as the board, and compiling, I get an error: "error: 'class String' has no member named 'isEmpty'"

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: Sequoia 15.2

**Additional context**
Add any other context about the problem here.

LynnL4, 2025-05-07:
Hi, I already fixed it
https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/commit/4fe22f26fd2b9bf0d5cdeee5b4ec279f22e65e69

limengdu, 2025-05-08:
The problem of compilation has been solved, but after uploading the program to Wio Terminal, no matter using uart or IIC, I can't get Wio Terminal to get the information about the recognition result of XIAO ESP32S3, and the serial port message is empty.

[image]

limengdu, 2025-05-22:
Can you please solve this issue as soon as possible? Anyone there?

LynnL4, 2025-05-22:
Could you try this fixed?
https://github.com/Seeed-Studio/Seeed_Arduino_SSCMA/commit/19601693f6c535806f4ea69f35a5e38481e81358

limengdu, 2025-05-23:
Thank you for your reply, the issue has been resolved.

[image]

closed 2025-05-23 by limengdu